JiriBilek/WiFiSpi #2: SSL support?

tonton81, 2017-06-11:
Most of the websites are switching to SSL, and that is VERY data hungry (crashing) for AT command set users, SPI should be plenty fast to keep up the buffer dequeuing

Is this library still in progress?

JiriBilek, 2017-06-11:
Yes, SSL support is high on my work list. The first implementation I wrote crashed the ESP, so more work is required.

tonton81, 2017-06-11:
ive gotten the AT command set working off an i2c uart adaptor @ 921600 baud with cts/rts and 64byte FIFOs, for regular use it's okay, but, SSL crashes the esp causing it to reboot, which is not so great. Hopefully the spi version can pull the data fast enough, you should have enough ram there to make a buffer for the spi requests

tonton81, 2017-06-11:
i've seen SSL bursts of like 40KB, the AT command set can only do 4-8KB max before crashing

JiriBilek, 2017-06-11:
I am afraid there would be a problem on the other side then. Arduino would be the weakest link.

tonton81, 2017-06-11:
arduino yes, but not teensy, which uses same code

tonton81, 2017-06-11:
if the arduino is doing parsing (not storing obviously) then it's feasible. I havn't thoughroly checked out your library, maybe your storing the read in an array before the user can access it, i dont know how the spi protocol is used here

JiriBilek, 2017-06-11:
The my Arduino part of the library can read arbitrary chunks of data. But the computing power of the Arduino is rather low so this could be the bottleneck. I'll know more after trying.

JiriBilek, 2020-12-29:
I am using the TLS connection without problems for some time, so I am closing it.
Nevertheless here is another issue connected to TLS: https://github.com/JiriBilek/WiFiSpi/issues/23

closed 2020-12-29 by JiriBilek