JiriBilek/WiFiSpi #3: WiFi shield not present on Mega2560 / Uno <-> Wemos D1 mini

ristomatti, 2017-12-10:
Hi,

I've tried wiring both a Mega2560 and Uno clones to a Wemos D1 mini according to the instructions but I'm getting `WiFi shield not present` on both the AVR boards on startup. The wiring is done through a logic level converter like this https://www.aliexpress.com/store/product/IIC-I2C-Logic-Level-Converter-Bi-Directional-Module-5V-to-3-3V-For-Arduino/612195_1893394721.html

There was no issues building or flashing on any of the boards and the D1 serial console says `SPI SLAVE ver. 0.1.1`. I've double checked the wirings and used all of the three boards to connect to miscellaneus SPI modules.

If I understood correctly from the code, WifiSpiESP has debug printing on by default. Is there a way to get more output on what the AVR side code is doing? I tried declaring `#define DEBUG` and `#define _DEBUG_` before `#include <WifiSpi.h>` but I'm not getting any more output.

OT background:

I came across your post at https://community.thinger.io/t/esp8266-and-arduino-connected-by-spi-interface/405 while searching to see if someone has implemented something like this. I've tried various serial based solutions to make these devices communicate nicely (https://github.com/bportaluri/WiFiEsp, https://github.com/jeelabs/esp-link and https://github.com/tuanpmt/esp_bridge for example), but either they were not working at all or failed at a specific use case I needed the connectivity for (mainly just MQTT publish & subscribe).

The most reliable solution for my particular use case has been code I wrote myself to forward MQTT data through serial as JSON (https://github.com/ristomatti/esp8266-rxtx-mqtt/blob/master/src/esp8266-rxtx-mqtt.cpp) but it's cumbersome to use as it requires separately flashing the ESP for all changes.

Out of frustration I yesterday started writing a more general purpose serial RPC callback scheme relying on https://github.com/thijse/Arduino-CmdMessenger for communication. This morning before continuing on the code I decided to do one more round of googling to see if there's some library/method I haven't run across and I ended up here full of hope as WifiSpi/WifiSpiESP looked to have a fresh approach. Hopefully the issue I'm having can be fixed and I can continue with writing some other code that requires a solid link between the devices! :)

ristomatti, 2017-12-10:
I guess it's worth mentioning I've used the logic level converter just recently for serial communication between the devices so I would doubt it to be the problem.

JiriBilek, 2017-12-10:
Hi,
are you sure you have proper wiring between Mega and ESP? The Mega SPI pins are not the same as for UNO.
See [https://www.arduino.cc/en/Reference/SPI](url)

My testing setup is NodeMCU and Pro Mini running on 5V. To convert voltage between them I use the same converter as you.

JiriBilek, 2017-12-10:
Posting a picture of my testing device. It was taken by my mobile phone so the quality is not very high.

[image]

JiriBilek, 2017-12-10:
And one more thought: have you connected grounds between Arduino, converter and Wemos?

tonton81, 2017-12-11:
be sure you use the github version of esp8266 for arduino as theres alot of features, fixes included in it thats not in the stock arduino release. it may ressolve the SPI slave issues your having

I was interested in the spi version but due to the bugs in the stock IDE it threw me to write UART version on both ends of the link, with 2 way acknowledgements, and 2-way control (multi-master mcu controller)

this gives complete uart control, i2c, spi control, gpio, analog, and user handler payload transfers all remotely, a type of setup that would make mesh networks cry, as the class objects can dynamically change winsocket ip depending on what you want to control. i plan to add more upgrades though
here is a demo of the project im working on, where 2 teensies at both ends of the link *attached to esp8266's) are controlling each other's led:
https://www.youtube.com/watch?v=RuPVjO1DDnY

but without the GIT version of esp8266, you'll have problems with both winsockets and SPI slave support

JiriBilek, 2017-12-11:
I didn't know about versions of esp8266 software for arduino. I am using the one from http://arduino.esp8266.com/stable/package_esp8266com_index.json

tonton81, 2017-12-11:
if thats for your boards manager its not the core
one sec, here: 
https://github.com/esp8266/Arduino/blob/master/README.md#using-git-version

the speed of the esp8266 also performs faster websockets connection/traffic than the one that ships with arduino IDE. keep in mind, the GIT has all updates and fixes that havnt been pushed into the IDE updates for quite a long time, uses latest lwip2 while the IDE is still on lwip1...

JiriBilek, 2017-12-11:
I see. I am generally not in favor to use nightly builds and I like to stick to stable versions (especially when my software works fine). But it is worth to try the github version.
Thanks for the information.

ristomatti, 2017-12-12:
@JiriBilek I have used separate SPI pins for Mega and the grounds were connected. I will check the wirings again just to be sure and post photos.

@tonton81 I read about the SPI slave issues when trying to debug this. I'll see if I'll try the GitHub version if I manage to get it running on PlatformIO. At least I was able to use the staging version of Arduino for ESP32.

Thank you both for replying!

ristomatti, 2017-12-12:
Let me know if the connections are difficult to see. It was surprisingly difficult to take a photo where you could clearly see all the connections.

[image]

I started thinking if the NodeMCU pin numbers are different than the ones on the Wemos D1 Mini. Here are the GPIO mappings from `pins_arduino.h`:

```c++
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
```

ristomatti, 2017-12-12:
Problem solved. I checked the connections the fifth time only to realize the level shifter has the 3.3V and 5V reversed! :unamused:.

I'm sorry for wasting your time! The issue can be closed. Awesome I'll get to properly test the library!

JiriBilek, 2017-12-13:
Oh, thanks for info. I am glad you found it.
Closing the issue.

closed 2017-12-13 by JiriBilek