JiriBilek/WiFiSpi #4: Use of other SPI interfaces?

CarlosSpin, 2017-12-11:
Hi,

Is there a way that I'm not seeing of using SPI1 or SPI2 interfaces instead of SPI without altering the library?

If not, could this be considered?

Best regards

JiriBilek, 2017-12-11:
Hi,
what hardware do you mean? Arduino Uno and Mega has AFAIK only one hardware SPI interface.

CarlosSpin, 2017-12-11:
Ah yes, sorry, forgot that information. I'm currently using a Teensy 3.6, which features three SPI interfaces.

JiriBilek, 2017-12-11:
I haven't used more SPI interface at one yet but I hope the only problems are:
- proper initialization on SPI object in your sketch (instead of using the default one in SPI.h)
- passing SS parameter to WiFiSpi.init(int8 pin);

Unfortunately, the library uses hardwired SPI variable so it wouldn't be straightforward to switch to SPI1, etc.

If you'd make any progress please share your knowledge.

tonton81, 2017-12-11:
yes it will work if you change SPI to SPI1 or SPI2.
edit all the required cpp files and you can do a mass replace like in wordpad on it (include the space)
" SPI." --> " SPI1"

jiribilek, if you decide to write multiple spi bus support, its real easy to add to the constructor, ive done it with my spi controller code to control 4 spi busses between 2 mcus :)

CarlosSpin, 2017-12-11:
Yes, I was searching for confirmation that there was no "hidden" way of changing that, without changing the library, that I was failing to see.

Since that is now stated, I am thinking of something more in line with @tonton81. I was actually thinking of doing the changes myself, as I need it right now, and submit a pull request, and leave it to you for further analysis.

JiriBilek, 2018-03-07:
Today, I started to test the library on STM32F1. I plan to add SPI2 and alternative pins for SPI1 soon (provided there is sufficient support for it in STM32duino).

CarlosSpin, 2018-03-09:
Unfortunately this had to wait for some time before I could address it. But this week I finally got around it. I have created a pull request that allows using any SPI interface by passing a reference to it, with the SPI (SPI0?!) as default.

closed 2018-03-09 by CarlosSpin