JiriBilek/WiFiSpi #6: recommended esp reset

jandrassy, 2018-02-19:
If a new (version of) sketch is uploaded or Serial Monitor resets the Arduino, the esp doesn't know and holds the unused connections.

If esp is reset and  the master holds the SS HIGH, the esp doesn't boot (we know why). It is sometimes hard to get out of this state.

I would recommend to WiFiSpi library users to connect the reset pin of esp8266 to some pin of Arduino (level shifted) and add this code to setup() of the sketch before WiFiSpi.init():
```
  pinMode(SS, OUTPUT);
  digitalWrite(SS, LOW);    // required boot state of the SS pin on esp8266

  pinMode(ESP_RESET_PIN, OUTPUT);
  digitalWrite(ESP_RESET_PIN, LOW); // low for reset
  delay(5);
  pinMode(ESP_RESET_PIN, INPUT); // let it to reset's pull-up circuit
  delay(1000); // wait for esp to boot

```
or support it in init() with optional espResetPin parameter

tonton81, 2018-02-19:
i did a port myself using 3 megabaud bidirectional uart and most of the core & libraries :)

tonton81, 2018-02-19:
esp-m3?
never heard of it :)
AsyncUDP, AsyncMQTT, AsyncWebsockets were ported to a non-wifi compliant mcu, the AT command set really sucks. I'm building a workhorse currently and have somewhere like 100 methods in a class that bring transparency (including handlers) over to it, with bauds running 2 000 000 - 3 000 000 with CRC checks as well. The beauty of the transparency is, if the core changes anything through updates, library is not affected. Almost all functions including their original returns are implemented. It's getting pretty crazy but I'm still working on it :)

JiriBilek, 2018-02-19:
M3: I bought a couple of them, they are based on ESP8255 but they are basically made for serial communication so I could not run the library on them. Just google. The firmware is identical to ESP8266.

Regarding the serial communication: I like the master-slave idea, for full duplex we'd need 4 channels. We can't mix commands and data. So it seemed to me that converting the library to serial communication would require constant polling of the ESP and I wasn't quite happy about it.

tonton81, 2018-02-19:
oh contrary, data goes in and out both ways
no polling, you send what you want, you get the response, if a callback occurs, it goes to the host!
i ported over all the handlers as well, including lambda anonymous one in async UDP
All packets are CRC verified and have yet to loose any at 2-3Megabaud, im even able to flood mqtt and receive async UDP packets at 0ms tight loop hammer mode :)

i just added yesturday callback queuing support
my handlers have 2 bonus features
1) non blocking (go ahead, sprinkle delays everywhere) :)
2) callbacks queued when functions are busy, and deque when free :) i call it “asychronous callback queuing” :)

whats better than a master-slave? multi-master!
the reason why the host is so updated is because ESP can talk to it too, as well as both ways can talk to each other. you can mix commands and data, ive done it, youd just have to see my videos for proof i guess, but id like to say, there is NO constant polling what im doing, how you poll is up to you with ESP commands on host, but there is NO polling in the library or UART communication

If you think thats crazy, I even ported over SPIFFS to the host. im able to read/write/create/append files on the ESP’s SPIFFS... from the host! :)

tonton81, 2018-02-19:
here i am flooding MQTT server with publishes in 25ms loops while receiving asynchronous UDP packets.
Ignore the DEC packets, it's my library debugging enabled while I work on it :)

async websockets while mqtt publishing, with UDP packets async https://www.youtube.com/watch?v=5H4WNRGchdI&t=7s

heck heres another video of me flooding an MQTT server at 0ms tight loops
https://www.youtube.com/watch?v=lpUslp-ZI44
what you see in the video is asynchronous callbacks of it receiving it back from subscription! :)

JiriBilek, 2018-02-19:
Thanks, I will look at it.

JiriBilek, 2018-02-19:
The speed is impressive. I admit, I don't understand the callback stuff you are writing about.

tonton81, 2018-02-19:
in serial monitor, you see the data which is from the callback, callbacks are for asynchronous events. lets take the stock ESP callbacks built in, ive ported them to the host. lets say wifi disconnects, or connects, or dhcp fails on connect, an event is fired on the ESP and it calls it’s core implemented callback. what i did basically is i create the identical callbacks, including their function arguments on the host end, and transfer the callback data over as a single payload which immediately is processed

tonton81, 2018-02-19:
if your familiar with the websocket callback on the ESP:

void webSocketEventServer(uint8_t num, WStype_t type, uint8_t * payload, size_t length) {

The exact one is implemented on the host.

If you get a websocket text, or message, the ESP handler forwards it to the HOST one instead
I basically gave transparency a whole new level by forwarding events and methods without modifications to libraries or core

tonton81, 2018-02-19:
I'd also like to mention no user code is stored on the ESP, ESP.* WiFi.* and libraries (WiFiClient/WiFiClientSecure,SPIFFS, EEPROM, UDP, websockets, etc etc etc), the host actually thinks it's an ESP at this point. Sorry no, I did not port over the SDK, that would be overkill :)

JiriBilek, 2018-02-19:
So kind of RPC?

tonton81, 2018-02-19:
RPC? now im lost :)

JiriBilek, 2018-02-19:
Sorry, https://en.wikipedia.org/wiki/Remote_procedure_call

tonton81, 2018-02-19:
yeah i guess it's similar to that
when I started designing it, it was suppose to be for MQTT client for mult-controller network, but thats been pushed aside temporarily once I kept upgrading the potential of this
The idea was to run the same functions locally as remotely, without any tampering of the code, which made it upgradable through core 2.3 -> 2.4 of the ESP, and still ongoing! the remote's stock function return goes back to the host and returns to user what the other end returned. This allowed me to port over tons of methods to make them run, including original naming scheme, as if the user was running an ESP (but really isn't). Everyday I look for more ways to advance it, it's getting bigger all the time :)

jandrassy, 2018-02-20:
@tonton81 you hijacked this issue. will you publish  your firmware and library?

tonton81, 2018-02-20:
eventually yeah, i already have a thread on a forum regarding the WIP, regarding your reset issue, GOIO15 must be low for a normal boot, which your using for SPI, and since your master doesn’t know if/when the ESP resets, thats where it gets stuck when the master is holding the line HIGH. you have no choice but to have access to reset gpio to reboot it to a known state as explined above. i used to use an additional gpio on the ESP (GPIO16) which would be driven high or low after sketch boot, to let the master know it was booted and ready, or completely lost (bad boot?), which then knew that it must be reset. i no longer do this as i use async uart for synchronization

JiriBilek, 2018-02-20:
@jandrassy Sorry, I am aware of the OT communication.
It is possible with external circuitry to avoid the situation when the ESP Reset does not work but I wanted to keep things as simple as possible.
See https://github.com/esp8266/Arduino/issues/2466#issuecomment-330322176
I filed an issue to the WiFiESP library. https://github.com/JiriBilek/WiFiSpiESP/issues/6

jandrassy, 2018-02-20:
@tonton81 I hade a plan to develop just for fun a proxy/stub/rpc system like you describe, so I am curios

tonton81, 2018-02-20:
im not sure on implementation of your idea as im not familiar with that method, but its good to have more ideas to expand it further. regarding my implementation, the functions do nothing but send the payload command, and retrieve the original return, before returning value to user, so the master code is basically, mostly, just uart data, with autosync and crc validations. bringing SPIFFS over was a major bonus. Theoretically the only way to ‘break’ the way the functions work is if the core/library authors decide to change the function overloads to a point where it affects their own users, which, you know, will never happen as no one wants to break an entire user base of people using their code. regardless, adding more methods or modifications is not that hard, but i keep it as transparent as possible to look and act as a real ESP as the objects and functions and callbacks are identical in every way

tonton81, 2018-02-20:
for example, my code doesnt just do WiFi.begin(“user”,”pass”);, surely no, im talking real transparency implementation, meaning you can input the additional channel, bssid, etc in the overloads and yeah, thats what i decided to work on

like ESP.restart(), funny thing is i added a lambda function of my own :) I called it ESP.onDetect[](){}; and you run it in setup with any of your ESP startup code, its binded to the ESP object so persists on master even after setup() exits, so example we drop WiFi.begin call in the onDetect lambda, if we call an ESP.reset as an example, the lambda will detect the ESP reconnect from callback and automatically resend the WiFi.begin call to it. I did a huge command call as a test and it works perfectly every time no matter how huge it is, even my callback implementations are non-blocking and support any commands you throw at it that wont work if you did that to an actual ESP

jandrassy, 2018-02-20:
@tonton81, you should learn some IT terminology to be able to communicate with other developers :-). 
Please publish on GitHub what you have or create a private repo with access for chosen (me :-)). 
I have test setups of esp8266's with Arduinos and experience with different esp8266 firmwares for arduino networking. I promise I will respect your code and coding style.

tonton81, 2018-02-20:
honestly i never used github so i wouldnt know where to start LOL, your a dev? yeah my lingo is not so good with words because im self taught over the last several years, sorry

jandrassy, 2018-02-20:
Yes I am a professional developer in Java and hobby developer for Arduino. Start with creating  a repo :-). You use Arduino IDE?

tonton81, 2018-02-20:
yeah always, and i use teensy mcus

tonton81, 2018-02-20:
my sense of direction to a project i started on that thread changed upwards of post #10 and continues to end which is progressing, so i put it aside to work on this as it was more demanding

https://forum.pjrc.com/threads/48839-New-project-code-named-teensquitto

tonton81, 2018-02-20:
the stale project will be remade in future to allow multi-network mcu controller support over mqtt using the library im working on which gives teensy as much benefits it can from the ESP, it should work on other mcus as well, but, provided the other micros have STL support as the async callback queue system uses std::deque with std::vectors in a push_back push_front scenario in order to keep the workflow solid. functional is also required to be included

tonton81, 2018-02-20:
pop_front i mean*

JiriBilek, 2018-03-09:
In development branch there is an implementation of soft reset. I mean reset over SPI. If you'd want to play with it don't forget to check out development branch on the WiFiSPiESP project, too.

JiriBilek, 2019-01-20:
Hard reset implemented in dev-0.2.0 branch by f5372495dd30ca6663edaeebf5df30759eb96323

JiriBilek, 2019-01-27:
Implemented in https://github.com/JiriBilek/WiFiSpi/commit/48ea9f39847ee7a302e850ab0ade206032bc41b9

closed 2019-01-27 by JiriBilek