JiriBilek/WiFiSpi #8: Random access time delay

CarlosSpin, 2018-03-09:
Using a Teensy 3.6 (STM32F4) I'm getting some random timeouts on SPI access. Using the changes in the Development branch severely reduces this effect, and fixes the random TCP disconnects. However, this still happens sometimes, making transmission of time sensitive information unreliable.

Also, I'm using an SPI connection of 10MHz, but I have also tried with 1MHz, and this still happens. What should be the maximum speed for the library ?

JiriBilek, 2018-03-09:
I'm investigating it. 
It is strange, looks like ESP processes a message coming on MOSI instead of transmitting via MISO. When a slave device transmits data, the master has to provide clock for it. This is done [here](https://github.com/JiriBilek/WiFiSpi/blob/fd38121450c747b4b3a34d1f3488ea47beb9b3bd/src/utility/espspi_proxy.h#L123). And ESP reads the bogus data instead of transmitting, seems to me.

CarlosSpin, 2018-03-09:
Ok, thanks ;)

Further debug shows that I'm still sometimes getting rejected messages due to invalid headers, even with 1MHz SPI connection.

tonton81, 2018-03-09:
teensy 3.6 is kinetis not stm... lol

CarlosSpin, 2018-03-09:
Yes, my bad, it uses an ARM Cortex M4F core, but is indeed a Kinetis.

JiriBilek, 2018-03-09:
I experienced the invalid messages on STM32F1 regardless what SPI speed I used. Most of them vanished when I added the SS delay on LOW signal before transmitting - [here](https://github.com/JiriBilek/WiFiSpi/blob/31da23deb271c15d11669164285f44e6034f06b2/src/utility/espspi_proxy.h#L89). But unfortunately not all of them.
I'd need to switch to old Arduino to see if the problem is related only to faster CPUs.

CarlosSpin, 2018-03-21:
Any development on this? I have been messing around with the library, and the WiFiSpiESP sketch, trying to establish a reliable source of communication between my ubuntu and the Teensy device, using the ESP. Everything works somewhat fine until I try to transfer a 1.5MB file from the Teensy SD card to the Ubuntu program. The file is transferred in 512 byte blocks, with all the transfer logic handled in the ubuntu and Teensy, which has already been verified using an USB connection.

A small summary of the transfer process is: teensy checks if esp client is connected, then reads the number of bytes available and, if any, reads all bytes avaliable. Then responds with the necessary information.

At a random point in the transfer process, I start getting timeouts in either the waitForSlaveRxReady or waitForSlaveTxReady functions of the library, together with timeouts in the readByte and flush functions of the WiFiSpiESP. A random combination of timeouts occurs, at any given step of the transfer.

JiriBilek, 2018-03-21:
Hi, have you tried the [development branch ](https://github.com/JiriBilek/WiFiSpi/tree/development) of the library?
If so, please post me a minimum app to reproduce the problem.
Thanks.

JiriBilek, 2019-01-27:
Commit https://github.com/JiriBilek/WiFiSpi/commit/48ea9f39847ee7a302e850ab0ade206032bc41b9 updated the protocol to a more robust one. If you are still interested, please try it. In case there are still problems, please open a new issue.

closed 2019-01-27 by JiriBilek