JiriBilek/WiFiSpi #9: Unstable

m0rt3nlund, 2018-06-25:
Hi!

I am still testing this and I am still having some problems with instability and I am unable to figure out what is the cause of this problems.

When I enable debugging in espspi_drv.cpp I get this in my serial: Cmd exp:A0, got:91
.status() command also returns WL_IDLE_STATUS and WL_NO_BOARD at random times.
Mostly WL_IDLE_STATUS.

The board is still connected like described in the issue https://github.com/JiriBilek/WiFiSpi/issues/5

Morten

m0rt3nlund, 2018-06-25:
Still using the example sketch WifiWebClient but added pinMode(PA4, OUTPUT); digitalWrite(PA4, LOW); to line 53 or I only get No shield Present at startup.

JiriBilek, 2018-06-29:
Hi,
have you tried the latest github version of ESP8266 arduino project? Just to have more info for debugging.
In my another project base pure on ESP it improved stability a lot.

Another try would be to slow down the SPI communication to be sure the problem is not in processing of SPI messages.

JiriBilek, 2018-10-31:
Hi, I have just comitted a important fix (https://github.com/JiriBilek/WiFiSpiESP/commit/836d098ecd3ed9a9891e826c63ea981521f4d45d) to the ESP8266 side (bad timing of MISO signal) that could solve the problems.
If you are still interested in the library I'd be glad if you test it, otherwise I'll close the issue.

m0rt3nlund, 2018-10-31:
I am still very interested! I will check it tomorrow! :-) Thank you for your work!

m0rt3nlund, 2018-11-04:
So far this is very promising! Good work! :) I Will also try to increase the speed as I have this limited to 1000000 at the moment for STM32.

Thank you for your good work! :D

JiriBilek, 2018-11-04:
Thanks,
on STM32F103 I ran 9Mhz clock, no problems :)

JiriBilek, 2019-01-27:
Closed as the protocol has changed recently (https://github.com/JiriBilek/WiFiSpi/commit/48ea9f39847ee7a302e850ab0ade206032bc41b9). If you are interested, please check the new version. If the problems persist, please open a new issue.

closed 2019-01-27 by JiriBilek