JiriBilek/WiFiSpi #14: Allow more than one connection in WiFiSPIServer

JiriBilek, 2019-01-06:
The WiFiSPIServer class allows only one incoming connection, although ESP8266 has no such limitation (in fact, there is obviously an limitation but is higher than one) . The limit is forced by the `_sock` array that serves as binding information between master and slave.
Discussed in #13

JiriBilek, 2019-01-12:
Strange, many internet sources say the ESP8266 can make 4 (or 5) connections at a time. I have just tested 32 concurrent TCPIP connections coming from 3 distinct IP addresses (not having more computers here).

m0rt3nlund, 2019-01-12:
I have read 8 maximum concurrent, but also that this is configurable :-)

Mvh
Morten Lund
+47 98203842

JiriBilek, 2019-01-13:
I did more tests - 16 concurrent clients running fine.
https://gist.github.com/JiriBilek/7bdde1863a3340ad58cbebbcc080583e