JiriBilek/WiFiSpi #16: WiFi module not present

kamalireza, 2019-05-28:
Hi, I compiled WiFiSpiESP(WiFiSPIESP.ino) to my Nodemcu(LoLin) and also compiled WiFiSPI(MQTT_Publish.ino) to my Arduino Mega 2560. 
I repetitively got "WiFi module not present" in output. Do you have any idea whats wrong?

my wiring is like this :
**Name**: **Nodemcu** - **Arduino Mega**
 **SS**: D8 - 53
 **MOSI**: D7 - 51
 **MISO**: D6 - 50
 **SCK**:  D5 - 52
3V arduino: Vin NodeMCU
GND arduino: Gnd NodeMCU

JiriBilek, 2019-05-28:
It's hard to say what's wrong. Obviously, the two modules don't communicate :)

Attach serial monitor to the NodeMCU to see if it works. Connect it via USB to your computer but do not connect power (3V) between the modules, only the ground.
Also you can experiment with [debug options](https://github.com/JiriBilek/WiFiSpiESP/blob/018349de89ef63ff8f4d01792f9bfda29b4c9eca/WiFiSPIESP/SPICalls.h#L30) and see if there are error messages.

kamalireza, 2019-05-28:
I attached serial monitor to the Nodemcu and only I received message "SPI SLAVE ver. 0.2.3 Protocol ver. 0.2.3". Is it correct? or I have to see something else?
I also checked with USB and grand but still not working! :(

JiriBilek, 2019-05-28:
Yes, it is correct.
Let's check the ESP does get a message. Please uncomment both _DEBUG and _DEBUG_MESSAGES in SPICalls.h in ESP project and load it once more.

kamalireza, 2019-05-28:
I did it, But nothing show up :|
just previous message that I wrote.
6 month ago I used your library and codes(older version), It seems fine. but now I don't know whats wrong :(

JiriBilek, 2019-05-28:
That could mean that ESP is not getting any data on SPI. Please double check the connection

kamalireza, 2019-05-28:
I already have checked the connection, and in [SPI ](https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave/examples)example was fine:

If you have any suggestion I will be grateful.

JiriBilek, 2019-05-28:
Did you set debugging on master (arduino)? It is in the file config.h:

```
// Uncomment the following define if the safe reset circuit (https://github.com/JiriBilek/WiFiSpiESP/issues/6)
// is connected to SS pin on ESP8266. This allows the SS signal to behave exactly
// according SPI specification (goes low before transmission and high after)
//#define ESP8266_SAFE_RESET_IMPLEMENTED

// Allow debugging information to be print on Serial
#define ESPSPI_DEBUG_OPTION
```

This should write more info on your Arduino.

kamalireza, 2019-05-28:
Yes, Now I get some error : 
```
[espspi_proxy.h:300] W: Slave rx is not ready, status 15
[espspi_proxy.h:300] W: Slave rx is not ready, status 15
[espspi_proxy.h:329] W: Slave tx is not ready, status 15
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[wifispi_drv.cpp:262] W: Error waitResponse
WiFi shield not present
```

JiriBilek, 2019-05-28:
It really looks like failure on SPI bus. Do you have a logic analyzer or oscilloscope?

kamalireza, 2019-05-28:
No, I haven't.

JiriBilek, 2019-05-28:
We need to see if there are any data on the bus. I suggest you to write a simple sketch on the ESP8266 side that reads the three digital inputs (SS, CLK, MOSI) and writes them to the serial monitor as fast as possible. It would be also fine to slow down the SPI bus by
`WiFiSpi.init(-1, 100000);` in the Arduino sketch.

srolf, 2019-05-28:
@kamalireza Maybe a stupid Question, but did you used a level shifter between the mega (5V) and the esp (3.3V) to align the Signal voltage?

kamalireza, 2019-05-29:
> We need to see if there are any data on the bus. I suggest you to write a simple sketch on the ESP8266 side that reads the three digital inputs (SS, CLK, MOSI) and writes them to the serial monitor [...]

I did that, I receive some data from it. But still arduino Mega not working!
I did something else. I have changed arduino Mega with Uno. and it's work! but there is some issue I had, I got some error message : 
```
Attempting to connect to SSID: Reza
Connected to wifi
[espspi_proxy.h:329] W: Slave tx is not ready, status 0
[espspi_proxy.h:329] W: Slave tx is not ready, status 0
```
ssid name is Reza.

Do you have any ideas whats wrong with my arduino Mega?

kamalireza, 2019-05-29:
> @kamalireza Maybe a stupid question, but did you used a level shifter between the mega (5V) and the esp (3.3V) to align the Signal voltage?

No, I did not use anything for voltage. I connect directly to 3V.

JiriBilek, 2019-05-29:
I have one Arduino Mega here. I will try and let you know.
Nevertheless, I would recommend you to use level shifter between 5V and 3.3V modules as the ESP8266 is said to be not 5V tolerant.

srolf, 2019-05-29:
I think it's Need. I was testing the implementation in different Setup, even with 3.3V M0 and 5V Leonardo.

kamalireza, 2019-05-31:
> I think it's need. I was testing the implementation in different setups, even with 3.3V M0 and 5V Leonardo.

I don't know what is the problem. Please let me know if there is a mistake:
I connect the Vin from Nodemcu to Arduino voltage supplier 5V. Also in datasheet in Nodemcu written Vin voltage must be 5V - 10V.

JiriBilek, 2019-05-31:
We are not talking about VIN but about GPIOs. 
The GPIO pin should not be connected to voltage higher than ca 3.6V according to the manufacturer. But if you google the internet for "ESP8266 5V tolerant" you will get many stories about how the MCU survived the higher voltage.

aldolo69, 2020-05-31:
hi. i'm not able to use this project but the spislave demo in the last esp8266 arduino lib works. i had garbage in the communication from slave to master with all the tests done before the upgrade to the very latest 2.7.1. maybe they have found a solution.

JiriBilek, 2020-05-31:
Could you please be more specific? Your device, connection, sketch, serial output on ESP8266?

aldolo69, 2020-05-31:
esp8266 + nano. nano send and esp receive (many status request in monitor) but the answer is not received and nano report "unable to connect". in many other test done with code found here and there i had the same result. even the master-slave example in the esp8266 core did not work. in hspi_slave.c  the  SPI1C2 = (0x1<<SPIC2MISODM_S); line is now a little bit different

JiriBilek, 2020-05-31:
First of all, I'd double check the wiring. If I understand, you are not able to connect either with this library or with the master-slave example from the core. Be sure to connect grounds between the two modules)

aldolo69, 2020-05-31:
done. it seem ok now. thanks.

closed 2020-05-31 by JiriBilek