JiriBilek/WiFiSpi #17: Protocol version mismatch. Please upgrade the firmware

johankson, 2019-08-11:
When flashing the ESP (built from master) it will be flashed with version 0.2.3 of the protocol. Installing the latest library for Arduino (Tools -> Library Manager) will get you version 0.2.0. This causes a "protocol version mismatch. Please upgrade the firmware" error.

Two issues
* The firmware is later than the Arduino client (downgrade the firmware?)
* There is no version 0.2.3 available for the Arduino client, do I need to build it myself?

JiriBilek, 2019-08-11:
You're right, sorry. I forgot to make a new release. The git version is up to date but Arduino IDE library management downloads released code only.
Fixed.

johankson, 2019-08-13:
Thanks for the quick response!

closed 2019-08-13 by johankson