JiriBilek/WiFiSpi #31: Protocol version mismatch. Please upgrade the firmware

yazeedpro, 2021-04-17:
Hi there

I hope this is not something that has been addressed elsewhere, but i am facing a little problem using this software

I have downloaded the software for the ESP8266 and the software for the Arduino, it seems that there is a version mismatch, when i go to grab matching versions, I find that even when both are labeled as release 0.2.5,  WiFiSpi contains version 0.2.3 and WiFiSpiESP is at 0.2.4, and I can not seem to find matching versions in both projects (At least not recent ones)

I am presented with the error "Protocol version mismatch. Please upgrade the firmware"

Thank you in advance

JiriBilek, 2021-04-17:
Please use the lastest github versions, they are synchronized. I am afraid, the releases are out of sync.

JiriBilek, 2021-04-18:
Hi, I checked the releases and protocol in both releases 0.2.5 match. Nevertheless in Arduino IDE, the recent version of WiFiSPI is 0.2.4 that won't work. 
As I am not familiar with the Arduino library updates, all I can suggest you is to get the release 0.2.5 from this repository, unpack it and overwrite the old version.

jandrassy, 2021-04-18:
Jiri, retag 2.5.0 with version 2.5.0 in library.properties

JiriBilek, 2021-04-18:
Juraj, thanks! With your help, I fixed it.

JiriBilek, 2021-04-19:
The change is propagated in Arduino IDE now.

closed 2021-04-19 by JiriBilek

yazeedpro, 2021-05-01:
Thank you jandrassy and Jiri, it works now ! have a blessed day ahead

yazeedpro, 2021-05-16:
Hello JiriBliek

The same issue appeared again after the 0.3.0 update
Once more, I am presented with the error "Protocol version mismatch. Please upgrade the firmware"

Thank you for all the effort you have put in this project
Kindly Advise

Yours
Yazeed

yazeedpro, 2021-05-16:
I apologize, My bad, it seems the upload to the Arduino failed, and hence the error which is obviously not an issue with this software