JiriBilek/WiFiSpi #32: Connect fail using BearSSL branch

nopnop2002, 2021-05-09:
I flashed WiFiSpiESP BearSSL branch to WeMos D1 Mini(ESP12).

And I opened https://github.com/JiriBilek/WiFiSpi/tree/BearSSL/examples/ConnectWithWPA, edit ssid & pass, flashed to Arduino Uno.

And I connected D1 and UNO, Reset Uno.

But connection fail.

Do you have any help.

- This is UNO side.
```
Attempting to connect to WPA SSID: aterm-d5a4ee-g 
[espspi_proxy.h:219] W: Bad CRC, retransmitting
Attempting to connect to WPA SSID: aterm-d5a4ee-g 
Attempting to connect to WPA SSID: aterm-d5a4ee-g 
Attempting to connect to WPA SSID: aterm-d5a4ee-g 
[espspi_proxy.h:219] W: Bad CRC, retransmitting
Attempting to connect to WPA SSID: aterm-d5a4ee-g 
Attempting to connect to WPA SSID: aterm-d5a4ee-g 
```

- This is ESP12 side.
```
Heap: 51784
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
<< df e0 50  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 7e
>> df e0 d0  1  5 30 2e 32 2e 35 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
<< dc e0 11  2  f 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67 20  d 37 35 33 30 63 34 38 37 32 62 19
<< df 34 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 fb
Conn: aterm-d5a4ee-g mode : sta(cc:50:e3:08:1c:d6)
add if0
 -> 7
>> df e0 91  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
scandone
no aterm-d5a4ee-g  found, reconnect after 1s
reconnect
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
<< df e0 50  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 7e
>> df e0 d0  1  5 30 2e 32 2e 35 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
<< dc e0 11  2  f 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67 20  d 37 35 33 30 63 34 38 37 32 62 19
<< df 34 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 fb
Conn: aterm-d5a4ee-g scandone
del if0
usl
mode : null
mode : sta(cc:50:e3:08:1c:d6)
add if0
 -> 7
>> df e0 91  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
scandone
no aterm-d5a4ee-g  found, reconnect after 1s
reconnect
Heap: 50656
scandone
no aterm-d5a4ee-g  found, reconnect after 1s
reconnect
scandone
no aterm-d5a4ee-g  found, reconnect after 1s
reconnect
<< dc e0 11  2  f 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67 20  d 37 35 33 30 63 34 38 37 32 62 19
<< df 34 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 fb
Conn: aterm-d5a4ee-g scandone
del if0
[... 155 lines cut ...]
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
Heap: 50656
scandone
no aterm-d5a4ee-g  found, reconnect after 1s
reconnect
scandone
no aterm-d5a4ee-g  found, reconnect after 1s
reconnect
scandone
no aterm-d5a4ee-g  found, reconnect after 1s
reconnect
<< dc e0 11  2  f 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67 20  d 37 35 33 30 63 34 38 37 32 62 19
<< df 34 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 fb
Conn: aterm-d5a4ee-g scandone
del if0
usl
mode : null
mode : sta(cc:50:e3:08:1c:d6)
add if0
 -> 7
>> df e0 91  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
```

nopnop2002, 2021-05-09:
Could the buffer be too small?

<< dc e0 11  2  f 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67 20  d 37 35 33 30 63 34 38 37 32 62 19

JiriBilek, 2021-05-09:
The buffer should be ok, the messages can be of (almost) any length. But 'Bad CRC' could show bad SPI connection.
I am not sure if you need 5V/3V3 convertor, but I'd try to put it in the SPI bus. And I'd try to slow down the SPI speed in case of any interference.

nopnop2002, 2021-05-09:
I'm sorry, I made a mistake on the UNO side.

wrong:
char ssid[] = "aterm-d5a4ee-g ";     // your network SSID (name)

correct:
char ssid[] = "aterm-d5a4ee-g";     // your network SSID (name)

But i can't connect.

```
SPI SLAVE ver. 0.2.5
Protocol ver. 0.2.5
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a 
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
<< df e0 50  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 7e 
>> df e0 d0  1  5 30 2e 32 2e 35 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
<< dc e0 11  2  e 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67  d 37 35 33 30 63 34 38 37 32 62 34 b8 
<< df 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 cc 
Conn: aterm-d5a4ee-gmode : sta(cc:50:e3:08:1c:d6)
add if0
ssid: [aterm-d5a4ee-g]passphrase: [7530c4872b4f2] -> 7
>> df e0 91  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a 
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with aterm-d5a4ee-g, channel 1
dhcp client start...
ip:192.168.10.133,mask:255.255.255.0,gw:192.168.10.1
Heap: 49880
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a 
>> df e0 a0  1  1  3 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
<< df e0 50  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 7e 
>> df e0 d0  1  5 30 2e 32 2e 35 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
<< dc e0 11  2  e 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67  d 37 35 33 30 63 34 38 37 32 62 34 b8 
<< df 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 cc 
Conn: aterm-d5a4ee-gstate: 5 -> 0 (0)
rm 0
Disconnect status=7
del if0
usl
mode : null
mode : sta(cc:50:e3:08:1c:d6)
add if0
ssid: [aterm-d5a4ee-g]passphrase: [7530c4872b4f2] -> 7
>> df e0 91  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a 
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with aterm-d5a4ee-g, channel 1
dhcp client start...
ip:192.168.10.133,mask:255.255.255.0,gw:192.168.10.1
```

JiriBilek, 2021-05-09:
I hope I understand the 'state' value well, then you are losing the connection.
The 'state: x->y' should write values from the following enum:
```
typedef enum {
    WL_NO_SHIELD        = 255,   // for compatibility with WiFi Shield library
    WL_IDLE_STATUS      = 0,
    WL_NO_SSID_AVAIL    = 1,
    WL_SCAN_COMPLETED   = 2,
    WL_CONNECTED        = 3,
    WL_CONNECT_FAILED   = 4,
    WL_CONNECTION_LOST  = 5,
    WL_DISCONNECTED     = 6
} wl_status_t;
```

Please try a sketch from Arduino/ESP8266 examples first in order to be sure there are no problems with your ESP8266 and it can connecti to the AP. 
Also I'd recommend to turn off the ESPSPI_MONITOR debugging when you urgently don't need it as it may break the SPI communication. From what I've seen, I can tell the communication between the Arduino and the ESP is ok.

nopnop2002, 2021-05-10:
I am investigating the cause.

My result:
```
<< dc e0 11  2  e 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67  d 37 35 33 30 63 34 38 37 32 62 34 b8 
cmdSetPassphrase
<< df 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 cc 
Conn: aterm-d5a4ee-gstate: 5 -> 0 (0)
rm 0
Disconnect status=7
del if0
usl
mode : null
mode : sta(cc:50:e3:08:1c:d6)
add if0
ssid: [aterm-d5a4ee-g]passphrase: [7530c4872b4f2] -> 7
>> df e0 91  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a 
cmdGetConnStatus
>> df e0 a0  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
scandone
state: 0 -> 2 (b0)
Heap: 49992
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with aterm-d5a4ee-g, channel 1
dhcp client start...
ip:192.168.10.133,mask:255.255.255.0,gw:192.168.10.1
```

Your expect:
```
<< dc e0 11  2  e 61 74 65 72 6d 2d 64 35 61 34 65 65 2d 67  d 37 35 33 30 63 34 38 37 32 62 34 b8 
cmdSetPassphrase
<< df 66 32 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 cc 
Conn: aterm-d5a4ee-gstate: 5 -> 0 (0)
rm 0
Disconnect status=7
del if0
usl
mode : null
mode : sta(cc:50:e3:08:1c:d6)
add if0
ssid: [aterm-d5a4ee-g]passphrase: [7530c4872b4f2] -> 7
>> df e0 91  1  1  7 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
scandone
state: 0 -> 2 (b0)
Heap: 49992
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 10
cnt 

connected with aterm-d5a4ee-g, channel 1
dhcp client start...
ip:192.168.10.133,mask:255.255.255.0,gw:192.168.10.1
<< df e0 20  0 ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 8a 
cmdGetConnStatus
>> df e0 a0  1  1  x ee  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
```

JiriBilek, 2021-05-10:
Hi, I tested ConnectWithWPA sketch on my device (STM32F103 + ESP8266) and got correct results (run several times to be sure it gives the same result for each run). 
I admit, I am not using the recent version of Arduino/ESP8266, I fixed the commit c61b70de865531f62218541fa0f32715c153a449 (14 months old) but it should not matter.

```
SDK:2.2.2-dev(a58da79)/Core:unspecified=0/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be

SPI SLAVE ver. 0.2.5
Protocol ver. 0.2.5
Conn: BILNetmode : sta(a0:20:a6:04:a5:8b)
add if0
 -> 6
scandone
state: 0 -> 2 (b0)
Heap: 40808
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with BILNet, channel 5
dhcp client start...
ip:192.168.1.183,mask:255.255.255.0,gw:192.168.1.254
Heap: 40000
pm open,type:2 0
Heap: 39480
Heap: 39480
```

nopnop2002, 2021-05-10:
I want to see your logs with _DEBUG_MESSAGES enabled.

nopnop2002, 2021-05-10:
>Edit: My SPI bus speed is 4 MHz.

Where is SPI SPEED defined?

JiriBilek, 2021-05-10:
The function `WiFiSpi.init();` has parameter `max_speed`.

nopnop2002, 2021-05-10:
If max_speed is not specified, UNO's SPI speed is probably 8Mhz.

If max_speed is specified, SPI.beginTransaction is executed.
But there is no SPI.endTransaction.

```
void WiFiSpiDrv::wifiDriverInit(uint8_t pin, uint32_t max_speed, SPIClass *in_spi)
{
    spi_obj = in_spi;
    spi_obj->begin();

    if (max_speed != 0)
        spi_obj->beginTransaction(SPISettings(max_speed, MSBFIRST, SPI_MODE0));
    
    espSpiProxy.begin(pin, spi_obj);
}
```

JiriBilek, 2021-05-10:
It's rather old code of mine, I can't remember all details, but AFAIK the endTransaction is not that important, try to slow down the bus.
IMHO, the Arduino bus is 4 MHz as well, see:
```
class SPISettings {
public:
  SPISettings(uint32_t clock, uint8_t bitOrder, uint8_t dataMode) {
    if (__builtin_constant_p(clock)) {
      init_AlwaysInline(clock, bitOrder, dataMode);
    } else {
      init_MightInline(clock, bitOrder, dataMode);
    }
  }
  SPISettings() {
    init_AlwaysInline(4000000, MSBFIRST, SPI_MODE0);
  }

```

JiriBilek, 2021-05-10:
I believe, beginTransaction is for us useful only in the two lines (AVR):
```
SPCR = settings.spcr;
SPSR = settings.spsr;
```

The SPI transaction in an Arduino construction, there no such term in SPI bus definition.

JiriBilek, 2021-05-10:
@nopnop2002 , I have just checked out the newest Arduino/ESP8266 libraries based on SDK 3.0.0. and it is failing here, too!!
So I assume the `SDK:2.2.2-dev(38a443e)/Core:unix-3.0.0-dev=-300000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df` introduced a bug in my app. I will look at it as the time allows, but for now, please check out an older version of Arduino/ESP8266 github project and give it a try.

nopnop2002, 2021-05-10:
>please check out an older version of Arduino/ESP8266 github project and give it a try.

Thank you for your help.

nopnop2002, 2021-05-11:
I tried new branch.
I got this error on Arduino UNO side.
Probably SPI communication is not established.

```
[espspi_proxy.h:300] W: Slave rx is not ready, status 15
[espspi_proxy.h:300] W: Slave rx is not ready, status 15
[espspi_proxy.h:329] W: Slave tx is not ready, status 15
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[espspi_proxy.h:261] W: Bad CRC, request repeated
[wifispi_drv.cpp:262] W: Error waitResponse
WiFi module not present
```

This is my compile option using WeMos D1 Mini.

[image]

JiriBilek, 2021-05-11:
Are you sure there is no issue with the SPI bus? My fix doesn't rely with the SPI bus at all.

JiriBilek, 2021-05-11:
I am not sure we are heading the right way now. The latest release in esp8266/Arduino repo **does not contain** the changed `wl_definitions.h` file.
So I'd suggest using the latest official esp8266/Arduino release for compiling and the issue would be gone. In future, when they make a new release, I'll update the library. Otherwise I am stuck in the problem of incompatibility.

JiriBilek, 2021-05-11:
Closing as the source of the problem lies outside this repo. I will update the library when the `wl_status_t` type is changed in the release of esp8266/Arduino repository. For now, this library works with last release of esp8266/Arduino lib, not with the git head.

nopnop2002, 2021-05-11:
Thank you again.