JiriBilek/WiFiSpi #34: Version mismatch

yazeedpro, 2021-05-16:
Hello JiriBilek

I have that same issue all over again, as soon as Arduino studio told me of the availability of 0.3.0, I updated both the library and the WiFiSPIESP, but the following problem showed up again.

"Protocol version mismatch. Please upgrade the firmware" showed up again.

Thank you for all the time and effort you put into this software, more power to you.

I have copied the old issue that was fixed before here for your referance

Yours
Yazeed

> Hi there

I hope this is not something that has been addressed elsewhere, but i am facing a little problem using this software

I have downloaded the software for the ESP8266 and the software for the Arduino, it seems that there is a version mismatch, when i go to grab matching versions, I find that even when both are labeled as release 0.2.5, WiFiSpi contains version 0.2.3 and WiFiSpiESP is at 0.2.4, and I can not seem to find matching versions in both projects (At least not recent ones)

I am presented with the error "Protocol version mismatch. Please upgrade the firmware"

Thank you in advance

yazeedpro, 2021-05-16:
Turned out the upload to the Arduino failed,  All is good, My bad, i apologize for rushing to report a non existent problem

closed 2021-05-16 by yazeedpro

JiriBilek, 2021-05-16:
No worries, I am fighting with the versions of both apps every time :)