JiriBilek/WiFiSpi #35: WiFiSpi.begin(...) problems with number of parameters

madmetz, 2022-09-02:
Hello Jiri,

After spending an entire day to find out why I can't get the ip address passed from my Mega2560 to my ESP8266 I was about to give up. But then I made a last attempt: What I found out is that the ip address is passed when I set all of the 4 parameters at once.
I.e. WifiSpi.begin(local_ip) doesn't work but WifiSpi.begin(local_ip, dns, gateway, subnet) works well.
I had a look at the WiFiSpi.h / WiFiSpi.cpp but I couldn't find out what goes wrong here.
Can you please have a look at this issue?

Best regards from Duesseldorf (Germany)

Michael

JAndrassy, 2022-09-02:
you mean WiFi.config

madmetz, 2022-09-02:
No, we are talking about WiFiSpi here - I thought. ;-)
Pls look at file WiFiSpi.h
But you are right - Of course I meant ".config" and not ".begin"
Thanx anyway.

JiriBilek, 2022-09-02:
Hi, I can't do any tests now. I'll be back in the middle of September.
Looking at the code I am afraid the code won't work without entering a gateway address. Precisely speaking, without specifying the gateway address, it is set to 0.0.0.0. I can't test it, I've only looked into the sources.

Look at: 

1. library WiFiSPI, file WiFiSPI.cpp, function `bool WiFiSpiClass::config(IPAddress local_ip)`
2. then app WiFiSPIESP, file WiFiSPICmdGeneral.cpp, function `void WiFiSpiEspCommandProcessor::cmdSetIpConfig()`
3. then library ESP8266WiFi, file ESP8266WIFISTA.cpp, function `bool ESP8266WiFiSTAClass::config(IPAddress local_ip, IPAddress arg1, IPAddress arg2, IPAddress arg3, IPAddress dns2)`
4. and finally <arduino_path>\hardware\esp8266\esp8266\tools\sdk\lwip\src\core\netif.c, function `void netif_set_addr(struct netif *netif, ip_addr_t *ipaddr, ip_addr_t *netmask, ip_addr_t *gw)`

If it were so, I'd have to delete the function `bool WiFiSpiClass::config(IPAddress local_ip)` from the library.

JAndrassy, 2022-09-03:
the Arduino's WiFiNINA and WiFi101 have the same error. the code to create the addresses is missing.
https://github.com/arduino-libraries/WiFiNINA/pull/219

madmetz, 2022-09-04:
@JiriBilek, @JAndrassy 
Hi and thanx for your replies.
I had a look at all the files and functions you listed and it's exactly the same on my machine.
Then I tried to find a workaround and decided to take my own ip address as gateway and dns like this:
  #ifdef IP_ADDRESS
    IPAddress local_ip(IP_ADDRESS);
    #ifdef DNS_ADDRESS
      IPAddress dns_server(DNS_ADDRESS);
    #else
      IPAddress dns_server(IP_ADDRESS);
    #endif
    #ifdef GW_ADDRESS
      IPAddress gateway(GW_ADDRESS);
    #else
      IPAddress gateway(IP_ADDRESS);
    #endif
    #ifdef SUBNET_MASK
      IPAddress subnet(SUBNET_MASK);
    #else   // if none specified we assume the smallest network class (C)
      IPAddress subnet(255, 255, 255, 0);
    #endif
  #endif

At least it works this way and I couldn't find any problems so far.
BTW: do you know if there is example code that makes the AVR (Mega) / ESP8266 work as a server that supports multiple sockets? The Ethernet library has a function "Ethernet.accept" which is missing in the WiFiSpi library.

JAndrassy, 2022-09-04:
> BTW: do you know if there is example code that makes the AVR (Mega) / ESP8266 work as a server that supports multiple sockets? The Ethernet library has a function "Ethernet.accept" which is missing in the WiFiSpi library.

my WiFiEspAT libary with Jiri's ATMod firmware