JiriBilek/WiFiSpi #36: Not able to connect secure server

codev123, 2024-07-20:
thank you sir for amazing work. I am using your library to send data to my https secure server, using POST method but it can not connect to https server.
```
Serial.println("Send weight to server :" + String(finalWeight));
  Serial.print("Starting connection to server...");
  Serial.println(centralServer);
  // if you get a connection, report back via serial:
  if (secureClient.connectSSL(centralServer, centralServerPort)) {
    Serial.println("connected to server");
    // Prepare the POST request
    String path = "/api/iotdata.php";
    String postData = "{\"device\":" + String(deviceID) + ",\"weight\":" + String(finalWeight, 2) + ",\"timestamp\":\"" + currentDateTime + "\"}";
    //final https request
    String request = String("POST ") + path + " HTTP/1.1\r\n" + "Host: " + String(centralServer) + "\r\n" + "User-Agent: Arduino/1.0\r\n" + "Connection: keep-alive\r\n" + "Content-Type: application/json\r\n" + "Content-Length: " + postData.length() + "\r\n" + "\r\n" + postData;
    //Serial.print("Http post request :");
    //Serial.println(request);
    // Convert String to char array
    char charRequest[request.length() + 1];
    request.toCharArray(charRequest, request.length() + 1);
    // Send the request
    secureClient.write(charRequest);
  }
```
while when i simply use 
```
if (secureClient.connect(centralServer, centralServerPort)) {
    Serial.println("connected to server"); }
```
i  am able to connect to server and send data to serevr. 
i an supply correct fingerprint and setting fingerprint too using
`WiFiSpi.setSSLFingerprint(fingerprint);`
please help me resolve this issue.

JiriBilek, 2024-07-20:
It's strange - I created a simple POST test using the code you had posted and it worked. I tested on STM32F103 a.k.a. Blue Pill, but you can use any arduino compatible CPU.
Try it:

```
/*
 * Testing POST request on httpbin.org
 */

#include "WiFiSpi.h"
#include "WiFiSpiClient.h"

#define WIFI_SSID "YOUR**SSID"
#define WIFI_PASSWORD "YOUR**PWD"

const char *host = "httpbin.org";
uint8_t fpHost[] = { 0x14,0x0C,0xC7,0xA8,0xEC,0xFA,0x7F,0x9C,0x9D,0xD2,0xB8,0x7E,0xC9,0xB8,0x93,0x3A,0xA1,0x11,0xF6,0x01 };

#define RESET_PIN PB12
#define SAMPLE_MILLIS (30 * 1000)

uint32_t lastTime = 0;

WiFiSpiClient client;

void setup() {
  // Press a key to start
  while (!Serial.available())
    ;

  Serial.println("Initializing");

  WiFiSpi.init(-1, 0, &SPI, RESET_PIN);  // init(int8_t pin = -1, uint32_t max_speed = 0, SPIClass *in_spi = &SPI, int8_t hwResetPin = -1);
}

void loop() {
  if (millis() - lastTime > SAMPLE_MILLIS || lastTime == 0) {
    if (WiFiSpi.status() != WL_CONNECTED) {
      Serial.println("Connecting to AP");
      WiFiSpi.begin(WIFI_SSID, WIFI_PASSWORD);
      if (WiFiSpi.status() == WL_CONNECTED)
        Serial.println("Connected to AP");
      else
        Serial.println("Cannot connect to AP");
    }

    lastTime = millis();

    Serial.print("Connecting to ");
    Serial.println(host);

    WiFiSpi.setSSLFingerprint(fpHost);

    if (client.connectSSL(host, 443) != 1) {
      Serial.println("Cannot connect to server");
    } else {
      Serial.println("Connected to server");
      
      String path = "/post";
      String postData = "{\"device\":1,\"weight\":42,\"timestamp\":\"112233\"}";
      //final https request
      String request = String("POST ") + path + " HTTP/1.1\r\n" + "Host: " + String(host) + "\r\n" + "User-Agent: Arduino/1.0\r\n" + "Connection: keep-alive\r\n" + "Content-Type: application/json\r\n" + "Content-Length: " + postData.length() + "\r\n" + "\r\n" + postData;
      char charRequest[request.length() + 1];
      request.toCharArray(charRequest, request.length() + 1);
      client.write(charRequest);

#define RESPONSE_TIMEOUT 5000UL
      unsigned long _startMillis = millis();
      while (millis() - _startMillis < RESPONSE_TIMEOUT && !client.available())
        ;

      while (client.available()) {
        char c = client.read();
        Serial.write(c);
      }

      Serial.println();

      client.stop();
    }
  }
}
```

codev123, 2025-02-20:
thank sir for your resonse, i have tried with your approch, even then i was not able to connect with connectSSL, 443 while able to conncet with connect, 80. 

`/*
 * Testing POST request on httpbin.org
 */

#include "WiFiSpi.h"
#include "WiFiSpiClient.h"

const int _MISO = 12;
const int _MOSI = 15;
const int _SCK = 14;
const int ESP8266_CS = 13;

#define RESPONSE_TIMEOUT 5000UL

#define WIFI_SSID "SSES"
#define WIFI_PASSWORD "Yes@080851"

const char *host = "httpbin.org";
uint8_t fpHost[] = { 0x14, 0x0C, 0xC7, 0xA8, 0xEC, 0xFA, 0x7F, 0x9C, 0x9D, 0xD2, 0xB8, 0x7E, 0xC9, 0xB8, 0x93, 0x3A, 0xA1, 0x11, 0xF6, 0x01 };

#define RESET_PIN PB12
#define SAMPLE_MILLIS (30 * 1000)

uint32_t lastTime = 0;

WiFiSpiClient secureClient;

void setup() {
  delay(2000);
  Serial.println("Initializing............");
  SPI1.setRX(_MISO);
  SPI1.setTX(_MOSI);
  SPI1.setSCK(_SCK);
  // Initialize the WifiSpi library
  WiFiSpi.init(ESP8266_CS, 4000000, &SPI1);
  // check for the presence of the shield:
  if (WiFiSpi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
  }
  if (!WiFiSpi.checkProtocolVersion()) {
    Serial.println("Protocol version mismatch. Please upgrade the firmware");
  }
  WiFiSpi.begin(WIFI_SSID, WIFI_PASSWORD);
  if (WiFiSpi.status() == WL_CONNECTED) {
    Serial.println("Connected to AP");
  } else {
    Serial.println("Cannot connect to AP");
  }
}

void loop() {
  if (millis() - lastTime > SAMPLE_MILLIS || lastTime == 0) {
    if (WiFiSpi.status() != WL_CONNECTED) {
      Serial.println("Connecting to AP");
      WiFiSpi.begin(WIFI_SSID, WIFI_PASSWORD);
      if (WiFiSpi.status() == WL_CONNECTED) {
        Serial.println("Connected to AP");
      } else {
        Serial.println("Cannot connect to AP");
      }
    }
    lastTime = millis();

    Serial.print("Connecting to ");
    Serial.println(host);

    WiFiSpi.setSSLFingerprint(fpHost);

    //if (secureClient.connectSSL(host, 443) != 1) {
    if (secureClient.connect(host, 80) != 1) {
      Serial.println("Cannot connect to server");
    } else {
      Serial.println("Connected to server");

      String path = "/post";
      String postData = "{\"device\":1,\"weight\":42,\"timestamp\":\"112233\"}";
      //final https request
      String request = String("POST ") + path + " HTTP/1.1\r\n" + "Host: " + String(host) + "\r\n" + "User-Agent: Arduino/1.0\r\n" + "Connection: keep-alive\r\n" + "Content-Type: application/json\r\n" + "Content-Length: " + postData.length() + "\r\n" + "\r\n" + postData;
      char charRequest[request.length() + 1];
      request.toCharArray(charRequest, request.length() + 1);
      secureClient.write(charRequest);
      unsigned long _startMillis = millis();
      while (millis() - _startMillis < RESPONSE_TIMEOUT && !secureClient.available())
        ;

      while (secureClient.available()) {
        char c = secureClient.read();
        Serial.write(c);
      }

      Serial.println();

      secureClient.stop();
    }
  }
}
`

codev123, 2025-02-20:
thank sir for your response, but with abov code too same issue existed, connect with 80 code and no connection with 443 or https

```
#include "WiFiSpi.h"
#include "WiFiSpiClient.h"

const int _MISO = 12;
const int _MOSI = 15;
const int _SCK = 14;
const int ESP8266_CS = 13;

#define RESPONSE_TIMEOUT 5000UL

#define WIFI_SSID "SSES"
#define WIFI_PASSWORD "Yes@080851"

const char *host = "httpbin.org";
uint8_t fpHost[] = { 0x14, 0x0C, 0xC7, 0xA8, 0xEC, 0xFA, 0x7F, 0x9C, 0x9D, 0xD2, 0xB8, 0x7E, 0xC9, 0xB8, 0x93, 0x3A, 0xA1, 0x11, 0xF6, 0x01 };

#define RESET_PIN PB12
#define SAMPLE_MILLIS (30 * 1000)

uint32_t lastTime = 0;

WiFiSpiClient secureClient;

void setup() {
  delay(2000);
  Serial.println("Initializing............");
  SPI1.setRX(_MISO);
  SPI1.setTX(_MOSI);
  SPI1.setSCK(_SCK);
  // Initialize the WifiSpi library
  WiFiSpi.init(ESP8266_CS, 4000000, &SPI1);
  // check for the presence of the shield:
  if (WiFiSpi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
  }
  if (!WiFiSpi.checkProtocolVersion()) {
    Serial.println("Protocol version mismatch. Please upgrade the firmware");
  }
  WiFiSpi.begin(WIFI_SSID, WIFI_PASSWORD);
  if (WiFiSpi.status() == WL_CONNECTED) {
    Serial.println("Connected to AP");
  } else {
    Serial.println("Cannot connect to AP");
  }
}

void loop() {
  if (millis() - lastTime > SAMPLE_MILLIS || lastTime == 0) {
    if (WiFiSpi.status() != WL_CONNECTED) {
      Serial.println("Connecting to AP");
      WiFiSpi.begin(WIFI_SSID, WIFI_PASSWORD);
      if (WiFiSpi.status() == WL_CONNECTED) {
        Serial.println("Connected to AP");
      } else {
        Serial.println("Cannot connect to AP");
      }
    }
    lastTime = millis();

    Serial.print("Connecting to ");
    Serial.println(host);

    WiFiSpi.setSSLFingerprint(fpHost);

    //if (secureClient.connectSSL(host, 443) != 1) {
    if (secureClient.connect(host, 80) != 1) {
      Serial.println("Cannot connect to server");
    } else {
      Serial.println("Connected to server");

      String path = "/post";
      String postData = "{\"device\":1,\"weight\":42,\"timestamp\":\"112233\"}";
      //final https request
      String request = String("POST ") + path + " HTTP/1.1\r\n" + "Host: " + String(host) + "\r\n" + "User-Agent: Arduino/1.0\r\n" + "Connection: keep-alive\r\n" + "Content-Type: application/json\r\n" + "Content-Length: " + postData.length() + "\r\n" + "\r\n" + postData;
      char charRequest[request.length() + 1];
      request.toCharArray(charRequest, request.length() + 1);
      secureClient.write(charRequest);
      unsigned long _startMillis = millis();
      while (millis() - _startMillis < RESPONSE_TIMEOUT && !secureClient.available())
        ;

      while (secureClient.available()) {
        char c = secureClient.read();
        Serial.write(c);
      }

      Serial.println();

      secureClient.stop();
    }
  }
}
`
```

codev123, 2025-02-20:
Thank you sir for yur response. but same issue still happening. 

#include "WiFiSpi.h"
#include "WiFiSpiClient.h"

const int _MISO = 12;
const int _MOSI = 15;
const int _SCK = 14;
const int ESP8266_CS = 13;

#define RESPONSE_TIMEOUT 5000UL

#define WIFI_SSID "SSID"
#define WIFI_PASSWORD "password"

const char *host = "httpbin.org";
uint8_t fpHost[] = { 0x14, 0x0C, 0xC7, 0xA8, 0xEC, 0xFA, 0x7F, 0x9C, 0x9D, 0xD2, 0xB8, 0x7E, 0xC9, 0xB8, 0x93, 0x3A, 0xA1, 0x11, 0xF6, 0x01 };

#define SAMPLE_MILLIS (30 * 1000)

uint32_t lastTime = 0;

WiFiSpiClient secureClient;

void setup() {
  delay(2000);
  Serial.println("Initializing............");
  SPI1.setRX(_MISO);
  SPI1.setTX(_MOSI);
  SPI1.setSCK(_SCK);
  // Initialize the WifiSpi library
  WiFiSpi.init(ESP8266_CS, 4000000, &SPI1);
  // check for the presence of the shield:
  if (WiFiSpi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
  }
  if (!WiFiSpi.checkProtocolVersion()) {
    Serial.println("Protocol version mismatch. Please upgrade the firmware");
  }
  WiFiSpi.begin(WIFI_SSID, WIFI_PASSWORD);
  if (WiFiSpi.status() == WL_CONNECTED) {
    Serial.println("Connected to AP");
  } else {
    Serial.println("Cannot connect to AP");
  }
}

void loop() {
  if (millis() - lastTime > SAMPLE_MILLIS || lastTime == 0) {
    if (WiFiSpi.status() != WL_CONNECTED) {
      Serial.println("Connecting to AP");
      WiFiSpi.begin(WIFI_SSID, WIFI_PASSWORD);
      if (WiFiSpi.status() == WL_CONNECTED) {
        Serial.println("Connected to AP");
      } else {
        Serial.println("Cannot connect to AP");
      }
    }
    lastTime = millis();

    Serial.print("Connecting to ");
    Serial.println(host);

    WiFiSpi.setSSLFingerprint(fpHost);

    //if (secureClient.connectSSL(host, 443) != 1) {        //not able to connect server
    if (secureClient.connect(host, 80) != 1) {                   //able to connect to server
      Serial.println("Cannot connect to server");
    } else {
      Serial.println("Connected to server");

      String path = "/post";
      String postData = "{\"device\":1,\"weight\":42,\"timestamp\":\"112233\"}";
      //final https request
      String request = String("POST ") + path + " HTTP/1.1\r\n" + "Host: " + String(host) + "\r\n" + "User-Agent: Arduino/1.0\r\n" + "Connection: keep-alive\r\n" + "Content-Type: application/json\r\n" + "Content-Length: " + postData.length() + "\r\n" + "\r\n" + postData;
      char charRequest[request.length() + 1];
      request.toCharArray(charRequest, request.length() + 1);
      secureClient.write(charRequest);
      unsigned long _startMillis = millis();
      while (millis() - _startMillis < RESPONSE_TIMEOUT && !secureClient.available())
        ;

      while (secureClient.available()) {
        char c = secureClient.read();
        Serial.write(c);
      }

      Serial.println();

      secureClient.stop();
    }
  }
}
`

closed 2025-02-20 by codev123